greearb/ath10k-ct #178: Firmware crash on UniFi AC Mesh

kroeckx, 2021-02-20:
I've recently upgraded my UniFi AC Mesh to OpenWRT 21.02 snapshot r15831-c74df745fd

After running a while, I had the following in dmesg:
```
[    0.000000] Linux version 5.4.99 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r15831-c74df745fd)) #0 Fri Feb 19 12:19:59 2021
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is Ubiquiti UniFi AC Mesh
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 122248K/131072K available (5223K kernel code, 189K rwdata, 680K rodata, 1196K init, 205K bss, 8824K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from 0x806559fc with crng_init=0
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008227] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
[    0.046713] pid_max: default: 32768 minimum: 301
[    0.051712] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.059427] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.071402] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.081711] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.088988] pinctrl core: initialized pinctrl subsystem
[    0.096879] NET: Registered protocol family 16
[    0.124232] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.137800] clocksource: Switched to clocksource MIPS
[    0.144258] NET: Registered protocol family 2
[    0.149699] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.158577] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.166665] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.174104] TCP: Hash tables configured (established 1024 bind 1024)
[    0.180928] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.187864] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.195504] NET: Registered protocol family 1
[    0.200158] PCI: CLS 0 bytes, default 32
[    0.207852] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[    0.220863] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.227050] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.250194] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.261419] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.268498] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.277381] printk: console [ttyS0] disabled
[    0.281968] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.291073] printk: console [ttyS0] enabled
[    0.300130] printk: bootconsole [early0] disabled
[    0.319582] spi-nor spi0.0: mx25l12805d (16384 Kbytes)
[    0.324990] 7 fixed-partitions partitions found on MTD device spi0.0
[    0.331564] Creating 7 MTD partitions on "spi0.0":
[... 326 lines cut ...]
[ 3703.128158] ath10k_pci 0000:00:00.0: [04]: 0x00058400   3   3  43   3
[ 3703.134827] ath10k_pci 0000:00:00.0: [05]: 0x00058800  18  18  17  18
[ 3703.141492] ath10k_pci 0000:00:00.0: [06]: 0x00058c00  27  27  27  27
[ 3703.148160] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[ 3703.154900] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x411ac0  dropped: 0
[ 3703.162743] ath10k_pci 0000:00:00.0: [0] next: 0x411aa8 buf: 0x40fefc sz: 1500 len: 144 count: 9 free: 0
[ 3703.172642] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 3703.179045] ath10k: [0000]: 27073A00 204CFC0B 0A000000 0B000000 3B073A00 204CFC0B 0A000000 0B000000
[ 3703.188435] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 3703.196056] ath10k: [0008]: 50073A00 204CFC0B 0A000000 0B000000 64073A00 204CFC0B 0A000000 0B000000
[ 3703.205562] ath10k: [0016]: 79073A00 204CFC0B 0A000000 0B000000 8D073A00 204CFC0B 0A000000 0B000000
[ 3703.214952] ath10k: [0024]: A2073A00 204CFC0B 0A000000 0B000000 B6073A00 204CFC0B 0A000000 0B000000
[ 3703.224308] ath10k: [0032]: CB073A00 204CFC0B 0A000000 0B000000
[ 3703.230431] ath10k_pci 0000:00:00.0: ATH10K_END
[ 3703.235144] ath10k_pci 0000:00:00.0: [1] next: 0x411ac0 buf: 0x40f90c sz: 1500 len: 0 count: 0 free: 0
[ 3703.250372] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer da376718 vdev: 0 addr: b4:fb:e4:75:27:0b 
[ 3703.363733] ieee80211 phy0: Hardware restart was requested
[ 3704.305809] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[ 3704.323776] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[ 3704.330254] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[ 3704.338570] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 24984 iram: 38672'
[ 3704.411044] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[ 3704.426473] ath10k_pci 0000:00:00.0: rts threshold -1
[ 3704.440636] ath10k_pci 0000:00:00.0: device successfully recovered
```

SimonIremonger, 2021-02-20:
I can confirm getting much the same errors, firmware crash, reset, etc... using a BT-HomeHub-v5-Type-A with the same OpenWRT 21.02SNAPSHOT with same ath10k firmware CT as well.
I also have older logs from running older 'master' build of OpenWRT from January without issue I believe.
I would like to provide more info / collect more debugging from console(serial)-log and syslog, if I know what debugging to turn on and what to get // what would be worthwhile....

greearb, 2021-02-21:
I cannot make progress on firmware crashes that are for this reason:
[ 3702.764066] ath10k_pci 0000:00:00.0: Cannot communicate with firmware, previous wmi cmds: 40859:849904 36904:849785 36904:849780 36904:849775, jiffies: 850688, attempting to fake crash and restart firmware, dev-flags: 0x42
The firmware stopped responding to the driver, so driver fakes a crash to recover.  That is likely as good as I can do.  In case those wmi-cmds are always the same in your crashes, just possibly they provide a clue, but I have dug through that code for weeks at a time with not much progress over the years...

kroeckx, 2021-02-21:
Note that my dmesg contains 2 such crashes:
```
ath10k_pci 0000:00:00.0: Cannot communicate with firmware, previous wmi cmds: 36904:353754 36954:353752 36904:353749 36952:353584, jiffies: 354560, attempting to fake crash and restart firmware, dev-flags: 0x42
ath10k_pci 0000:00:00.0: Cannot communicate with firmware, previous wmi cmds: 40859:849904 36904:849785 36904:849780 36904:849775, jiffies: 850688, attempting to fake crash and restart firmware, dev-flags: 0x42
```

They don't seem to have anything in common.

I've now switched that device to non-ct, and it works without problems so far.

kroeckx, 2021-02-21:
Should any firmware version work with the driver I have? If @SimonIremonger can say which firmware version worked I can try to do a bisect, but I guess it will be slow since it's takes time to reproduce.

kroeckx, 2021-02-21:
I might have misunderstand what's in this git repo. I now think it's only the driver, not the firmware.

greearb, 2021-02-21:
The wave-1 CT firmware is based on a pretty old upstream firmware, and all of it is closed source.  If QCA firmware works better for you then, that is probably best.  The ath10k-ct driver should work fine with upstream QCA firmware, but the stock ath10k driver may work fine for you as well.

klukonin, 2021-02-22:
@greearb I see a possible approach to switch ath10k-ct beacon mode to "BURST". This mode should (I guess) avoid some timing or queuing issues in firmware.

SimonIremonger, 2021-05-13:
Informal reports indicate issues here occurring into OpenWRT 21.02 rc (imminent release...) versions, informed users needing to switch to non -ct firmware.  Will try to add some serial logging and debugging/monitoring to my ath10k-containing gadgets...

greearb, 2021-05-13:
The -ct firmware for wave-2 is completely different beast from the wave-1 -ct firmware.  Please be specific about what platform(s) and/or chipsets seem to have issues or not.

CodeFetch, 2021-07-26:
@greearb From what I know the UniFi AC Mesh has a rebranded QCA9886. Isn't that wave 2? Is it possible the wrong firmware was chosen for it or something?

greearb, 2021-07-26:
9886 is wave-2.  9888 aka 988x is wave-1.

greearb, 2021-09-22:
The original bug report was about wave-1, radio times out, and I don't know how to fix.  The driver did recover the system after the radio became un-responsive.  Closing this bug

closed 2021-09-22 by greearb