greearb/ath10k-ct #179: Set tx-antenna pattern on a per-peer basis

francescamen, 2021-03-24:
I'm trying to use the functionality to set tx-antenna pattern on a per-peer basis.
My hardware has 4 antennas and I want to assign two of them to one specific peer with id 1.
As stated in the instructions, if I want to set peer-id 0x1 to use antenna pattern 0xc I have to do the following:
`# echo 0x130001000c > ct_special`
However, when I check if the set has been executed correctly, looking at the dmesg output, I see the following:
`ath10k_pci 0000:01:00.0: Setting ct-andmask for peer: 1 to 0x4.`
I don't know if the problem is related to hardware constraints or it is only a problem of the print. 
In that respect, I'm wondering why, in the file debug.c, the value to be printed is masked with 0x16. I would mask it with 0x1F instead.
`ath10k_warn(ar, "Setting ct-andmask for peer: %d to 0x%x.\n", val >> 16, val & 0x16);`
Is this the problem?
Thank you.

Hardware: Netgear R7800 
NIC chipset: QCA9984
Firmware version 10.4b-ct

greearb, 2021-09-22:
Thank for debugging this, I think you are right and this is just a printing error.  Were you able to see any actual change in behaviour when you set the peer antenna mask like this?

greearb, 2021-09-22:
I added the fix to my 5.10 and higher drivers, easy enough to add to older ones if someone cares to.  It is cosmetic issue anyway.

closed 2021-09-22 by greearb