greearb/ath10k-ct #182: Debian 10.9, 5.10.24-1~bpo10+1 kernel, dmesg error, NIC not up

atzlinux, 2021-04-26:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Software (OS, Firmware version, kernel, driver, etc)
OS: Debian 10.9
Kernel: 
5.10.0-0.bpo.5-rt-amd64 #1 SMP PREEMPT_RT Debian 5.10.24-1~bpo10+1 (2021-03-29) x86_64 GNU/Linux
Firmware version:
QCA9984-firmware-5-ct-full-community-12.bin-lede.022

This firmware is download from http://sources.openwrt.org/.

Hardware (NIC chipset, platform, etc)
lspci -v

07:00.0 Network controller: Qualcomm Atheros QCA9984 802.11ac Wave 2 Wireless Network Adapter
	Subsystem: Qualcomm Atheros QCA9984 802.11ac Wave 2 Wireless Network Adapter
	Flags: bus master, fast devsel, latency 0, IRQ 165
	Memory at a1000000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable+ Count=1/32 Maskable+ 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Kernel driver in use: ath10k_pci
	Kernel modules: ath10k_pci

Logs (dmesg, maybe supplicant and/or hostap)

[ 2606.216437] ath10k_core: loading out-of-tree module taints kernel.
[ 2606.216695] ath10k_core: module verification failed: signature and/or required key missing - tainting kernel
[ 2618.742493] ath10k 5.10 driver, optimized for CT firmware, probing pci device: 0x46.
[ 2618.743533] ath10k_pci 0000:07:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[ 2618.854003] ath10k_pci 0000:07:00.0: firmware: failed to load ath10k/fwcfg-pci-0000:07:00.0.txt (-2)
[ 2618.854014] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[ 2618.854040] ath10k_pci 0000:07:00.0: firmware: failed to load ath10k/pre-cal-pci-0000:07:00.0.bin (-2)
[ 2618.854065] ath10k_pci 0000:07:00.0: firmware: failed to load ath10k/cal-pci-0000:07:00.0.bin (-2)
[ 2618.854092] ath10k_pci 0000:07:00.0: firmware: failed to load ath10k/QCA9984/hw1.0/ct-firmware-5.bin (-2)
[ 2618.854116] ath10k_pci 0000:07:00.0: firmware: failed to load ath10k/QCA9984/hw1.0/ct-firmware-2.bin (-2)
[ 2618.854139] ath10k_pci 0000:07:00.0: firmware: failed to load ath10k/QCA9984/hw1.0/firmware-6.bin (-2)
[ 2618.854871] ath10k_pci 0000:07:00.0: firmware: direct-loading firmware ath10k/QCA9984/hw1.0/firmware-5.bin
[ 2618.854890] ath10k_pci 0000:07:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[ 2618.854898] ath10k_pci 0000:07:00.0: kconfig debug 0 debugfs 0 tracing 0 dfs 0 testmode 0
[ 2618.856297] ath10k_pci 0000:07:00.0: firmware ver 10.4b-ct-9984-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7ea63dc5
[ 2620.125523] ath10k_pci 0000:07:00.0: firmware: direct-loading firmware ath10k/QCA9984/hw1.0/board-2.bin
[ 2620.125609] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:31 crc32 85498734
[ 2622.599850] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[ 2622.599856] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[ 2622.675532] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[ 2622.676199] ath10k_pci 0000:07:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
[ 2622.862332] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[ 2623.020058] ath: EEPROM regdomain sanitized
[ 2623.020064] ath: EEPROM regdomain: 0x64
[ 2623.020067] ath: EEPROM indicates we should expect a direct regpair map
[ 2623.020070] ath: Country alpha2 being used: 00
[ 2623.020072] ath: Regpair used: 0x64
[ 2623.020390] ------------[ cut here ]------------
[ 2623.020392] WARNING: CPU: 3 PID: 24327 at net/wireless/core.c:621 wiphy_register+0x486/0x9f0 [cfg80211]
[ 2623.020502] Modules linked in: ath10k_pci(OE) ath10k_core(OE) ath mac80211 libarc4 fuse snd_hda_codec_hdmi joydev snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_sof_xtensa_dsp snd_sof snd_sof_intel_hda ledtrig_audio snd_soc_skl snd_soc_hdac_hda x86_pkg_temp_thermal snd_hda_ext_core intel_powerclamp snd_soc_sst_ipc cfg80211 coretemp snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi kvm_intel rfkill snd_hda_intel kvm snd_intel_dspcfg 8021q soundwire_intel irqbypass garp soundwire_generic_allocation stp snd_soc_core mrp crc32_pclmul llc ghash_clmulni_intel snd_compress soundwire_cadence aesni_intel libaes crypto_simd snd_hda_codec cryptd glue_helper snd_hda_core rapl snd_hwdep evdev intel_rapl_msr intel_cstate soundwire_bus iTCO_wdt intel_pmc_bxt snd_pcm i915 iTCO_vendor_support intel_uncore serio_raw sg pcspkr watchdog ee1004 snd_timer wmi_bmof intel_wmi_thunderbolt snd soundcore drm_kms_helper cec processor_thermal_device mei_me intel_rapl_common drm
[ 2623.020603]  int340x_thermal_zone mei intel_soc_dts_iosf intel_pch_thermal intel_pmc_core acpi_pad acpi_tad button ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic uas usb_storage hid_generic usbhid hid sd_mod t10_pi crc_t10dif crct10dif_generic ahci xhci_pci libahci xhci_hcd igb sdhci_pci i2c_algo_bit libata cqhci crct10dif_pclmul dca intel_lpss_pci crct10dif_common sdhci ptp usbcore i2c_i801 psmouse intel_lpss crc32c_intel scsi_mod pps_core i2c_smbus usb_common mmc_core idma64 fan wmi video
[ 2623.020712] CPU: 3 PID: 24327 Comm: kworker/u16:1 Tainted: G           OE     5.10.0-0.bpo.5-rt-amd64 #1 Debian 5.10.24-1~bpo10+1
[ 2623.020719] Hardware name: Default string Default string/Default string, BIOS 5.13 06/17/2020
[ 2623.020723] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[ 2623.020771] BUG: using smp_processor_id() in preemptible [00000000] code: kworker/u16:1/24327
[ 2623.020776] caller is print_stop_info+0x1b/0x40
[ 2623.020787] CPU: 3 PID: 24327 Comm: kworker/u16:1 Tainted: G           OE     5.10.0-0.bpo.5-rt-amd64 #1 Debian 5.10.24-1~bpo10+1
[... 42 lines cut ...]
[ 2623.021612] R13: 0000000000000020 R14: 0000000000000002 R15: 0000000000000001
[ 2623.021615] FS:  0000000000000000(0000) GS:ffff8a6325cc0000(0000) knlGS:0000000000000000
[ 2623.021619] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2623.021623] CR2: 00007f26841a57f8 CR3: 0000000196810001 CR4: 00000000003706e0
[ 2623.021627] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 2623.021629] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 2623.021632] Call Trace:
[ 2623.021636]  ? __rtnl_unlock+0x25/0x40
[ 2623.021643]  ? netdev_run_todo+0x5e/0x2d0
[ 2623.021652]  ieee80211_register_hw+0x682/0xbe0 [mac80211]
[ 2623.021739]  ? ath10k_regd_update+0x2a0/0x2a0 [ath10k_core]
[ 2623.021772]  ath10k_mac_register+0xa7d/0xea0 [ath10k_core]
[ 2623.021810]  ath10k_core_register_work+0x6f0/0xc70 [ath10k_core]
[ 2623.021848]  ? __schedule+0x2b8/0x890
[ 2623.021854]  process_one_work+0x1a1/0x3d0
[ 2623.021865]  worker_thread+0x30/0x370
[ 2623.021874]  ? process_one_work+0x3d0/0x3d0
[ 2623.021882]  kthread+0x183/0x1a0
[ 2623.021889]  ? kthread_park+0x80/0x80
[ 2623.021896]  ret_from_fork+0x1f/0x30
[ 2623.021908] ---[ end trace 0000000000000002 ]---
[ 2624.209158] ath10k_pci 0000:07:00.0: failed to register ieee80211: -22
[ 2624.209172] ath10k_pci 0000:07:00.0: could not register to mac80211 (-22)

NIC is not up, iwconfig can't see any wireless NIC.

gsustek, 2021-12-17:
apply this ! https://github.com/greearb/ath10k-ct/blob/master/patches/0100-mac80211-fix-txq-queue-related-crashes.patch
to your kernel

atzlinux, 2021-12-18:
Thanks for your feedback.

I don't use this NIC now, so I can't to test it.

You may close this issue.

greearb, 2021-12-18:
I think the problem is that my driver will not work properly in un-patched upstream kernel.  You can clone one of the kernels from my site in that case, or dig up whatever patch in my kernel fixes that assert.

closed 2021-12-18 by greearb