greearb/ath10k-ct #183: Firmware is crashing on tp-link router

bttd, 2021-05-04:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Hi,

This problem happening every 2 or 3 times a week. We using macbooks connected to the router.
The firmware crashing sometimes, and every device lose the network connection until the wifi restarts. 

Software (OS, Firmware version, kernel, driver, etc)

Hostname | OpenWrt
Firmware Version | OpenWrt 21.02.0-rc1 r16046-59980f7aaf / LuCI openwrt-21.02 branch git-21.106.55967-06dd6b5
Kernel Version | 5.4.111

Same happened on stable OpenWrt (19.07.7)

Hardware (NIC chipset, platform, etc)

Model | TP-Link Archer C7 v5
Architecture | Qualcomm Atheros QCA956X ver 1 rev 0

Logs (dmesg, maybe supplicant and/or hostap)

`[    0.000000] Linux version 5.4.111 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16046-59980f7aaf)) #0 Sun Apr 18 10:06:57 2021
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C7 v5
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 122244K/131072K available (5216K kernel code, 190K rwdata, 684K rodata, 1196K init, 205K bss, 8828K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from 0x80655984 with crng_init=0
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008229] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
[    0.046713] pid_max: default: 32768 minimum: 301
[    0.051715] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.059423] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.071659] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.081946] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.089231] pinctrl core: initialized pinctrl subsystem
[    0.097073] NET: Registered protocol family 16
[    0.124757] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.138238] clocksource: Switched to clocksource MIPS
[    0.144734] NET: Registered protocol family 2
[... 843 lines cut ...]
[313387.060421] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[313387.060435] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.060442] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.060484] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[313387.060498] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.060504] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.068237] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[313387.068250] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.068257] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.088521] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.088531] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.088562] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.088568] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.109321] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.109331] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.109357] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.109363] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.122027] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.122037] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.129771] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.129781] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.136382] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.136392] ath10k_pci 0000:00:00.0: failed to push frame: -19
[313387.200267] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[313387.200277] ath10k_pci 0000:00:00.0: failed to push frame: -19`

greearb, 2021-05-04:
The firmware hung, which is unfortunately somewhat normal, and then driver forced restart:
[313384.521735] ath10k_pci 0000:00:00.0: Cannot communicate with firmware, previous wmi cmds: 36904:78270981 36954:78270754 36904:78270743 36954:78270721, jiffies: 78271552, attempting to fake crash and restart firmware, dev-flags: 0x42
[313384.543466] ath10k_pci 0000:00:00.0: failed to send wmi nop: -143
[313384.550012] ath10k_pci 0000:00:00.0: could not request stats (type -268435456 ret -143 specifier 1)

Did system recover automatically after this, or did you have to reboot it?

bttd, 2021-05-05:
System recover automatically in few minutes

clebersantz, 2021-05-24:
I have the same problem, with CT firmware sometime WIFI freezes and need to restart router to get it working again.
Using non CT firmware it doesn't occurs and I get faster speeds.

This happens with 2.4GHz, on a TP-Link Archer C7 v2 with OpenWRT 21.02, using ath10k-firmware-qca988x-ct and kmod-ath10k-ct.

There as lot of topics in openwrt about it even with others OpenWRT versions, but none solution. See https://forum.openwrt.org/t/archer-c7-2-4-ghz-wireless-dies-in-24-48-hours/.

there is a workaround that works for me as well, when running the command below the problem still occurs, but WIFI continue to work after that:
$ iw dev wlan_2g scan trigger freq 2462 flush

greearb, 2021-09-22:
@clebersantz seems you have a different bug, your system does not actually recover with ath10k-ct driver/firmware (sometimes?)  If you can get dmesg output after a freeze (and that may require serial console), then I will see if I can notice anything interesting.  Please open a new bug if you want to pursue that.  And, what problem still occurs but is different when you do that scan trigger freq thing?  Closing this bug...plz add answers/comments to new bug.

closed 2021-09-22 by greearb