greearb/ath10k-ct #184: Question: Support for encap offload in tx

Ansuel, 2021-05-12:
@greearb I hope I'm not bothering you. The qsdk have this patch that never actually took part to the ath10k upstream but the firmware actually supports it. With some quick test, this patch gives 100+ mb of extra band (normal ath10k base + normal ath10k firmware + the encap offlocat patch) with the same cpu usage.
Trying to adapt the same path with the ath10k-ct code and a firmware full firmware (that should have all the feature compiled in) cause a firmware crash as soon as some packet are transmitted.

So the question is:
Does your firmware supports this mode? It seems a good thing and I think anything that can give better perf should not be ignored or at least compiled in the firmware. Do you have more info or any idea why the firmware crash with this?

Thanks in advance for any answer to this question.

castiel652, 2021-05-12:
FYI this was the patch that was close to be accepted.
https://patchwork.kernel.org/project/linux-wireless/patch/20191216092207.31032-1-john@phrozen.org/

and this one is what I use in my personal build but I made some change to adapt to the mac80211 change. 
https://github.com/castiel652/openwrt/commit/26b7cf2065cb6372ae2aaeb5edf0a4f865622fa9

greearb, 2021-05-12:
Please attach the firmware crash log.

Ansuel, 2021-05-12:
```
[  389.017520] ath10k_pci 0001:01:00.0: Invalid peer id 0 or peer stats buffer, peer: 201eb761  sta: 00000000
[  390.618912] ath10k_pci 0000:01:00.0: firmware crashed! (guid c4d87147-b907-43d0-b64a-501cfd13a0a9)
[  390.619017] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[  390.626949] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  390.642246] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-tH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b8ee13f
[  390.654830] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 85498734
[  390.675470] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[  390.683661] ath10k_pci 0000:01:00.0: firmware register dump:
[  390.692777] ath10k_pci 0000:01:00.0: [00]: 0x0000000A 0x00000000 0x0098BEF7 0x00000000
[  390.699224] ath10k_pci 0000:01:00.0: [04]: 0x00000000 0x00060024 0x00000000 0x00000000
[  390.706333] ath10k_pci 0000:01:00.0: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.714328] ath10k_pci 0000:01:00.0: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.722203] ath10k_pci 0000:01:00.0: [16]: 0x00981D37 0x009606B8 0x009606CA 0x0098BEF7
[  390.730120] ath10k_pci 0000:01:00.0: [20]: 0x00000000 0x00401C10 0x00000000 0x00000000
[  390.738010] ath10k_pci 0000:01:00.0: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.745831] ath10k_pci 0000:01:00.0: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.754372] ath10k_pci 0000:01:00.0: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.761721] ath10k_pci 0000:01:00.0: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.770169] ath10k_pci 0000:01:00.0: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.777495] ath10k_pci 0000:01:00.0: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.785312] ath10k_pci 0000:01:00.0: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.793816] ath10k_pci 0000:01:00.0: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.801209] ath10k_pci 0000:01:00.0: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[  390.809645] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[  390.817021] ath10k_pci 0000:01:00.0: [00]: 0x0004a000   4   4   3   3
[  390.822721] ath10k_pci 0000:01:00.0: [01]: 0x0004a400  10  10 208 209
[  390.830293] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  43  43  42  43
[  390.835738] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  21  21  25  21
[  390.843104] ath10k_pci 0000:01:00.0: [04]: 0x0004b000  56  56  70  30
[  390.848690] ath10k_pci 0000:01:00.0: [05]: 0x0004b400   5   5  68  69
[  390.855009] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  28  28  28  28
[  390.862398] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
[  390.867966] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
[  390.874368] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   1   1   1   1
[  390.881691] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
[  390.887209] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
[  390.893603] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x420ed8  dropped: 0
[  390.900930] ath10k_pci 0000:01:00.0: [0] next: 0x420ec0 buf: 0x417820 sz: 1500 len: 796 count: 30 free: 0
[  390.907631] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  390.917499] ath10k: [0000]: 00004A46 1400587C 51100001 000EC9FC 000003FC 00000004 00450454 00004A46
[  390.923064] ath10k: [0008]: 1000587A 0042CCF4 00450454 01000000 00000001 00004A46 14006403 1B4D0730
[  390.932095] ath10k: [0016]: 6CEE0000 00000001 00000000 00450454 00004A49 10005881 00003112 0042CCF4
[  390.941123] ath10k: [0024]: 0000000A 00000000 00004A49 10005881 00003112 0042CCF4 0000000E 00000000
[  390.950622] ath10k: [0032]: 00004A4A 10004C01 71100D08 00000000 0000000A 00000077 00004A4A 14004C01
[  390.959180] ath10k: [0040]: 71104441 030AEE6C 0601B003 0000FFFA 0000FFFF 00004A4A 13FC4C01 000081DF
[  390.968630] ath10k: [0048]: 00457371 00000008 00000060 00004A4A 14004C01 71108880 00000000 00D80060
[  390.977237] ath10k: [0056]: 00000000 00000FF0 00004A4A 14004C01 71108880 00010000 00D80060 00000000
[  390.986170] ath10k: [0064]: 0000FFFF 00004A4A 14004C01 71108880 00020000 00D80060 00000000 0000FFFF
[  390.995296] ath10k: [0072]: 00004A4A 14004C01 71108880 00030000 00D80060 00000000 0007FDFF 00004A4A
[  391.004322] ath10k: [0080]: 14004C01 71108880 00040000 00D80060 00000000 0007FDFF 00004A4A 14004C01
[  391.013774] ath10k: [0088]: 71108880 00050000 00D80060 00000000 000FFFFF 00004A4A 14004C01 71108880
[  391.022378] ath10k: [0096]: 00060000 00D80060 00000000 00000000 00004A4A 14004C01 71104441 030AEE6C
[  391.031822] ath10k: [0104]: 0601B003 0000FFFA 0000FFFF 00004A4B 13FC4C01 000081DF 00457E6D 00000008
[  391.040435] ath10k: [0112]: 00000060 00004A4B 14004C01 71108880 00000000 00D80060 00000000 00000FF0
[  391.049878] ath10k: [0120]: 00004A4B 14004C01 71108880 00010000 00D80060 00000000 0000FFFF 00004A4B
[  391.058492] ath10k: [0128]: 14004C01 71108880 00020000 00D80060 00000000 0000FFFF 00004A4B 14004C01
[  391.067933] ath10k: [0136]: 71108880 00030000 00D80060 00000000 0007FDFF 00004A4B 14004C01 71108880
[... cut ...]
[  391.437679] ath10k_pci 0001:01:00.0: debug log header, dbuf: 0x420ed8  dropped: 0
[  391.443952] ath10k_pci 0001:01:00.0: [0] next: 0x420ec0 buf: 0x417820 sz: 1500 len: 0 count: 0 free: 0
[  391.452329] ath10k_pci 0001:01:00.0: [1] next: 0x420ed8 buf: 0x417230 sz: 1500 len: 0 count: 0 free: 0
[  391.507679] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer a9734cdb vdev: 0 addr: 30:07:4d:1b:ee:6c 
[  391.507756] ath10k_pci 0000:01:00.0: removing peer, cleanup-all, deleting: peer ef70c2e9 vdev: 0 addr: dc:ef:09:f2:5d:51 
```

greearb, 2021-05-12:
Please test with this, and I'll likely need the binary crash dump to get the debug out of the core.  There is some NPE early in the tx path, maybe it did not find the peer object.
[firmware-5-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/6467264/firmware-5-htt-mgt-community.bin.gz)

Ansuel, 2021-05-12:
@greearb sure... how can I provide you the binary crash dump ? should I enable some debug flag when I insmod the module?

greearb, 2021-05-12:
something like: cat /sys/kernel/debug/ieee80211/phy0/ath10k/fw_crash_dump > /tmp/fw_dump.bin
after the crash, and assuming 'phy0' is the radio that crashed.  You should get a reasonable sized binary
file, or an error if there is no crash dump available.

Ansuel, 2021-05-12:
@greearb Are you interested in the crashdump with the new firmware or do you want also the crashdump from the old one?

greearb, 2021-05-12:
the new one, I added equivalent of 'printk' near the crash site to try to understand it better.

greearb, 2021-05-12:
Please re-run with this one, it has more debugging.  Only the binary dump is useful for this type of debugging.
[firmware-5-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/6468558/firmware-5-htt-mgt-community.bin.gz)

Ansuel, 2021-05-12:
new:
https://we.tl/t-0MHnk40SLw

greearb, 2021-05-12:
I either gave you the wrong file, or you ran with the old firmware img.  Plz double-check, and if you are owrt-devel irc, ping me (greearb)

Ansuel, 2021-05-12:
@greearb i'm on the irc chat

Ansuel, 2021-05-13:
I you have any clue or any idea to test, feel free to write or ask here and I will join irc for any test.

nicklowe, 2021-05-23:
This would be really good to move forward to significantly improve the performance.

Ansuel, 2021-05-23:
The update on this is that the firmware crash is fixed but we have no connection...
The firmware part that should handle this is untested and we should find the reason of no traffic
I send some wifi dump but greearb is currently busy so we just need to wait and hope ;)

Ansuel, 2021-08-08:
@greearb any news on this? I can provide whatever data you need

Ansuel, 2021-09-23:
@greearb Any chance for this?

Ansuel, 2022-06-19:
@greearb considering encap offload support was merged upstream wonder if there is some chance of fixing support for it for your firmware
Happy to help you debug any problem with pcap packet and other stuff :D