greearb/ath10k-ct #193: linux 5.15.5 firmware-ct cause ath10k driver to crash

gsustek, 2021-11-28:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Software (OS, Firmware version, kernel, driver, etc)

Hardware (NIC chipset, platform, etc)

Logs (dmesg, maybe supplicant and/or hostap)

[Sat Nov 27 21:38:14 2021] ath10k 5.15 driver, optimized for CT firmware, probing pci device: 0x46.
[Sat Nov 27 21:38:14 2021] ath10k_pci 0000:07:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[Sat Nov 27 21:38:14 2021] ath10k_pci 0000:07:00.0: fwcfg key: regdom  val: 332
[Sat Nov 27 21:38:14 2021] ath10k_pci 0000:07:00.0: fwcfg key: dma_burst  val: 0
[Sat Nov 27 21:38:14 2021] ath10k_pci 0000:07:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[Sat Nov 27 21:38:14 2021] ath10k_pci 0000:07:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[Sat Nov 27 21:38:14 2021] ath10k_pci 0000:07:00.0: firmware ver 10.4b-ct-9984-xfH-13-b1b524c8e5 api 5 features mfp,peer-flow-ctrl,peer-fixed-rate,txstatus-noack,wmi-10.x-CT,rxswcrypt-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,CT-STA,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 f866180c
[Sat Nov 27 21:38:15 2021] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:31 crc32 e807b522
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: DANGER! You're overriding EEPROM-defined regulatory domain
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: from: 0x0 to 0x14c (svc-ready-work)
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: Your card was not certified to operate in the domain you chose.
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: This might result in a violation of your local regulatory rules.
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: Do not ever do this unless you really know what you are doing!
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: wmi print 'ChipType 1100 ChipTypeID 0 BoardID 2 P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: wmi print 'free: 85112 iram: 19908 sram: 12824'
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[Sat Nov 27 21:38:18 2021] ath: EEPROM regdomain: 0x814c
[Sat Nov 27 21:38:18 2021] ath: EEPROM indicates we should expect a country code
[Sat Nov 27 21:38:18 2021] ath: doing EEPROM country->regdmn map search
[Sat Nov 27 21:38:18 2021] ath: country maps to regdmn code: 0x37
[Sat Nov 27 21:38:18 2021] ath: Country alpha2 being used: HT
[Sat Nov 27 21:38:18 2021] ath: Regpair used: 0x37
[Sat Nov 27 21:38:18 2021] Modules linked in: ath10k_pci(O) ath10k_core(O) nfnetlink bridge stp llc ip6table_filter ip6_tables xt_MASQUERADE xt_nat iptable_nat nf_nat xt_TCPMSS iptable_mangle xt_limit ipt_REJECT nf_reject_ipv4 xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_comment iptable_filter dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua snd_soc_skl snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core ath snd_hda_codec_hdmi snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel x86_pkg_temp_thermal snd_intel_dspcfg intel_powerclamp kvm_intel snd_hda_codec snd_usb_audio snd_hda_core mac80211 kvm snd_usbmidi_lib iTCO_wdt snd_hwdep iTCO_vendor_support snd_pcm uvcvideo intel_rapl_msr irqbypass snd_seq_midi crct10dif_pclmul snd_seq_midi_event crc32_pclmul snd_rawmidi ghash_clmulni_intel processor_thermal_device_pci_legacy snd_seq videobuf2_vmalloc rapl processor_thermal_device videobuf2_memops processor_thermal_rfim intel_cstate
[Sat Nov 27 21:38:18 2021]  processor_thermal_mbox i2c_i801 videobuf2_v4l2 snd_seq_device processor_thermal_rapl snd_timer videobuf2_common intel_wmi_thunderbolt i2c_smbus wmi_bmof sdhci_pci serio_raw intel_rapl_common intel_lpss_pci igb videodev libarc4 snd uas mei_me int340x_thermal_zone cqhci intel_lpss usb_storage idma64 mc rtl8188fu(O) soundcore mei dca intel_soc_dts_iosf sdhci virt_dma intel_pch_thermal acpi_pad sch_fq_codel it87 hwmon_vid coretemp cuse msr parport_pc ppdev lp parport ip_tables x_tables autofs4 raid10 raid456 libcrc32c async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq raid1 raid0 multipath linear i915 cec rc_core i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm aesni_intel psmouse crypto_simd ahci cryptd libahci wmi video mac_hid [last unloaded: ath10k_core]
[Sat Nov 27 21:38:18 2021] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[Sat Nov 27 21:38:18 2021]  ? ath_reg_apply_world_flags+0x28/0xa0 [ath]
[Sat Nov 27 21:38:18 2021]  ath10k_mac_register+0xb51/0xf20 [ath10k_core]
[Sat Nov 27 21:38:18 2021]  ? ath10k_wmi_detach+0x3e/0x80 [ath10k_core]
[Sat Nov 27 21:38:18 2021]  ath10k_core_register_work+0x921/0xd30 [ath10k_core]
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: failed to register ieee80211: -22
[Sat Nov 27 21:38:18 2021] ath10k_pci 0000:07:00.0: could not register to mac80211 (-22)

greearb, 2021-11-28:
Did this system work on older versions?
Can you try with US regdomain instead of HT?

gsustek, 2021-11-28:
hi, it was working. i must warn you that my wificard has region 00 and and that is why during compilation of kernel i needed to use above stated kernel configuration.
i did not apply any patch to regd.c or kernel at all...

if i change HT to US:

[Sun Nov 28 20:34:14 2021] ath10k 5.15 driver, optimized for CT firmware, probing pci device: 0x46.
[Sun Nov 28 20:34:14 2021] ath10k_pci 0000:07:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[Sun Nov 28 20:34:14 2021] ath10k_pci 0000:07:00.0: fwcfg key: regdom  val: 840
[Sun Nov 28 20:34:14 2021] ath10k_pci 0000:07:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[Sun Nov 28 20:34:14 2021] ath10k_pci 0000:07:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[Sun Nov 28 20:34:14 2021] ath10k_pci 0000:07:00.0: firmware ver 10.4-ct-9984-xfH-011-33ec52c8 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,rxswcrypt-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 5b4f6cbc
[Sun Nov 28 20:34:16 2021] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:31 crc32 85498734
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: DANGER! You're overriding EEPROM-defined regulatory domain
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: from: 0x0 to 0x348 (svc-ready-work)
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: Your card was not certified to operate in the domain you chose.
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: This might result in a violation of your local regulatory rules.
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: Do not ever do this unless you really know what you are doing!
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: wmi print 'free: 90288 iram: 15972 sram: 19864'
[Sun Nov 28 20:34:18 2021] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[Sun Nov 28 20:34:19 2021] ath: EEPROM regdomain: 0x8348
[Sun Nov 28 20:34:19 2021] ath: EEPROM indicates we should expect a country code
[Sun Nov 28 20:34:19 2021] ath: doing EEPROM country->regdmn map search
[Sun Nov 28 20:34:19 2021] ath: country maps to regdmn code: 0x3a
[Sun Nov 28 20:34:19 2021] ath: Country alpha2 being used: US
[Sun Nov 28 20:34:19 2021] ath: Regpair used: 0x3a
[Sun Nov 28 20:34:19 2021] Modules linked in: ath10k_pci(O) ath10k_core(O) ccm nfnetlink bridge stp llc ip6table_filter ip6_tables xt_MASQUERADE xt_nat iptable_nat nf_nat xt_TCPMSS iptable_mangle xt_limit ipt_REJECT nf_reject_ipv4 xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_comment iptable_filter dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua snd_soc_skl snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core x86_pkg_temp_thermal intel_powerclamp snd_compress kvm_intel ath snd_hda_codec_hdmi ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg mac80211 snd_usb_audio snd_hda_codec kvm snd_hda_core snd_usbmidi_lib snd_hwdep snd_pcm uvcvideo irqbypass snd_seq_midi iTCO_wdt crct10dif_pclmul snd_seq_midi_event crc32_pclmul videobuf2_vmalloc iTCO_vendor_support intel_rapl_msr snd_rawmidi ghash_clmulni_intel videobuf2_memops snd_seq videobuf2_v4l2 rapl processor_thermal_device_pci_legacy intel_cstate videobuf2_common
[Sun Nov 28 20:34:19 2021]  processor_thermal_device snd_seq_device serio_raw intel_wmi_thunderbolt wmi_bmof processor_thermal_rfim videodev snd_timer processor_thermal_mbox i2c_i801 processor_thermal_rapl igb uas snd libarc4 i2c_smbus sdhci_pci mei_me intel_lpss_pci intel_rapl_common mc intel_lpss usb_storage rtl8188fu(O) int340x_thermal_zone cqhci soundcore idma64 mei dca intel_soc_dts_iosf sdhci intel_pch_thermal virt_dma mac_hid acpi_pad sch_fq_codel it87 hwmon_vid coretemp cuse msr parport_pc ppdev lp parport ip_tables x_tables autofs4 raid10 raid456 libcrc32c async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq raid1 raid0 multipath linear i915 cec rc_core i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm aesni_intel psmouse crypto_simd cryptd ahci libahci wmi video [last unloaded: ath10k_core]
[Sun Nov 28 20:34:19 2021] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[Sun Nov 28 20:34:19 2021]  ? ath_reg_apply_world_flags+0x28/0xa0 [ath]
[Sun Nov 28 20:34:19 2021]  ath10k_mac_register+0xb51/0xf20 [ath10k_core]
[Sun Nov 28 20:34:19 2021]  ? ath10k_wmi_detach+0x3e/0x80 [ath10k_core]
[Sun Nov 28 20:34:19 2021]  ath10k_core_register_work+0x921/0xd30 [ath10k_core]
[Sun Nov 28 20:34:19 2021] ath10k_pci 0000:07:00.0: failed to register ieee80211: -22
[Sun Nov 28 20:34:19 2021] ath10k_pci 0000:07:00.0: could not register to mac80211 (-22)
root@u1804:/lib/firmware/ath10k/QCA9984/hw1.0#

gsustek, 2021-11-28:
this is without any regdomain value:

[Sun Nov 28 20:37:04 2021] ath10k 5.15 driver, optimized for CT firmware, probing pci device: 0x46.
[Sun Nov 28 20:37:04 2021] ath10k_pci 0000:07:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[Sun Nov 28 20:37:04 2021] ath10k_pci 0000:07:00.0: loading /lib/firmware/ath10k/fwcfg-pci-0000:07:00.0.txt failed with error -22
[Sun Nov 28 20:37:04 2021] ath10k_pci 0000:07:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[Sun Nov 28 20:37:04 2021] ath10k_pci 0000:07:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[Sun Nov 28 20:37:04 2021] ath10k_pci 0000:07:00.0: firmware ver 10.4-ct-9984-xfH-011-33ec52c8 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,rxswcrypt-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT crc32 5b4f6cbc
[Sun Nov 28 20:37:05 2021] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:31 crc32 85498734
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: wmi print 'free: 90288 iram: 15972 sram: 19864'
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[Sun Nov 28 20:37:08 2021] ath: EEPROM regdomain sanitized
[Sun Nov 28 20:37:08 2021] ath: EEPROM regdomain: 0x64
[Sun Nov 28 20:37:08 2021] ath: EEPROM indicates we should expect a direct regpair map
[Sun Nov 28 20:37:08 2021] ath: Country alpha2 being used: 00
[Sun Nov 28 20:37:08 2021] ath: Regpair used: 0x64
[Sun Nov 28 20:37:08 2021] Modules linked in: ath10k_pci(O) ath10k_core(O) ccm nfnetlink bridge stp llc ip6table_filter ip6_tables xt_MASQUERADE xt_nat iptable_nat nf_nat xt_TCPMSS iptable_mangle xt_limit ipt_REJECT nf_reject_ipv4 xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_comment iptable_filter dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua snd_soc_skl snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core x86_pkg_temp_thermal intel_powerclamp snd_compress kvm_intel ath snd_hda_codec_hdmi ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg mac80211 snd_usb_audio snd_hda_codec kvm snd_hda_core snd_usbmidi_lib snd_hwdep snd_pcm uvcvideo irqbypass snd_seq_midi iTCO_wdt crct10dif_pclmul snd_seq_midi_event crc32_pclmul videobuf2_vmalloc iTCO_vendor_support intel_rapl_msr snd_rawmidi ghash_clmulni_intel videobuf2_memops snd_seq videobuf2_v4l2 rapl processor_thermal_device_pci_legacy intel_cstate videobuf2_common
[Sun Nov 28 20:37:08 2021]  processor_thermal_device snd_seq_device serio_raw intel_wmi_thunderbolt wmi_bmof processor_thermal_rfim videodev snd_timer processor_thermal_mbox i2c_i801 processor_thermal_rapl igb uas snd libarc4 i2c_smbus sdhci_pci mei_me intel_lpss_pci intel_rapl_common mc intel_lpss usb_storage rtl8188fu(O) int340x_thermal_zone cqhci soundcore idma64 mei dca intel_soc_dts_iosf sdhci intel_pch_thermal virt_dma mac_hid acpi_pad sch_fq_codel it87 hwmon_vid coretemp cuse msr parport_pc ppdev lp parport ip_tables x_tables autofs4 raid10 raid456 libcrc32c async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq raid1 raid0 multipath linear i915 cec rc_core i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm aesni_intel psmouse crypto_simd cryptd ahci libahci wmi video [last unloaded: ath10k_core]
[Sun Nov 28 20:37:08 2021] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[Sun Nov 28 20:37:08 2021]  ? ath_reg_apply_world_flags+0x28/0xa0 [ath]
[Sun Nov 28 20:37:08 2021]  ath10k_mac_register+0xb51/0xf20 [ath10k_core]
[Sun Nov 28 20:37:08 2021]  ? ath10k_wmi_detach+0x3e/0x80 [ath10k_core]
[Sun Nov 28 20:37:08 2021]  ath10k_core_register_work+0x921/0xd30 [ath10k_core]
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: failed to register ieee80211: -22
[Sun Nov 28 20:37:08 2021] ath10k_pci 0000:07:00.0: could not register to mac80211 (-22)

gsustek, 2021-11-28:
i even apply thi patch:
ath-reg: Relax application of default world regd on my 5.15.5 which change ath.ko, but no change...

gsustek, 2021-11-29:
it seems that it works now!

1.  git clone git://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack v5.15.5v
2. git reset --hard b10c942b042c318e4d7783cd529f0fd5f468eed6v
3. patch
4.  git apply -v ../ath10k-ct-5.15/patches/0001-wireless-Relax-beacon_int_min_gcd-and-ADHOC-check.patch
5.  git apply -v ../0404-ath-reg-Relax-application-of-default-world-regd.patch git apply -v ../0404-ath-reg-Relax-application-of-default-world-regd.patch
3. build kernel with those connfig (CONFIG_ATH_REG_DYNAMIC_USER_REG_HINTS=y CONFIG_ATH_REG_DYNAMIC_USER_CERT_TESTING=y CONFIG_ATH10K_DFS_CERTIFIED=y CONFIG_CFG80211_CERTIFICATION_ONUS=y)
It seems that is working for now!

but ping is fairly high.

64 bytes from 192.168.168.1: icmp_seq=7 ttl=64 time=14.129 ms
64 bytes from 192.168.168.1: icmp_seq=8 ttl=64 time=9.750 ms
64 bytes from 192.168.168.1: icmp_seq=9 ttl=64 time=10.296 ms
64 bytes from 192.168.168.1: icmp_seq=10 ttl=64 time=9.228 ms
64 bytes from 192.168.168.1: icmp_seq=11 ttl=64 time=9.820 ms
64 bytes from 192.168.168.1: icmp_seq=12 ttl=64 time=9.542 ms
64 bytes from 192.168.168.1: icmp_seq=13 ttl=64 time=5.475 ms
64 bytes from 192.168.168.1: icmp_seq=14 ttl=64 time=10.234 ms
64 bytes from 192.168.168.1: icmp_seq=15 ttl=64 time=3.903 ms
64 bytes from 192.168.168.1: icmp_seq=16 ttl=64 time=7.157 ms
64 bytes from 192.168.168.1: icmp_seq=17 ttl=64 time=3.219 ms

closed 2021-11-29 by gsustek