greearb/ath10k-ct #194: bad ping time on ubuntu vs openwrt

gsustek, 2021-11-29:
These are the steps how to build kernel for x86 machine with 9984 wifi in AP mode card.
OpenWRT and DDWRT has better performance then regular ubuntu 20.04 LTS distro.
The one who has full linux distro also as router, can you check your ping/packet loss?

git clone git://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack v5.15.5v
git reset --hard b10c942b042c318e4d7783cd529f0fd5f468eed6v
patch
git apply -v ../ath10k-ct-5.15/patches/0001-wireless-Relax-beacon_int_min_gcd-and-ADHOC-check.patch
git apply -v ../0404-ath-reg-Relax-application-of-default-world-regd.patch git apply -v ../0404-ath-reg-Relax-application-of-default-world-regd.patch
build kernel with those connfig (CONFIG_ATH_REG_DYNAMIC_USER_REG_HINTS=y CONFIG_ATH_REG_DYNAMIC_USER_CERT_TESTING=y CONFIG_ATH10K_DFS_CERTIFIED=y CONFIG_CFG80211_CERTIFICATION_ONUS=y)
It seems that is working for now!
but ping is fairly high.

64 bytes from 192.168.168.1: icmp_seq=662 ttl=64 time=12.231 ms
64 bytes from 192.168.168.1: icmp_seq=663 ttl=64 time=9.781 ms
64 bytes from 192.168.168.1: icmp_seq=664 ttl=64 time=2.943 ms
64 bytes from 192.168.168.1: icmp_seq=665 ttl=64 time=8.656 ms
64 bytes from 192.168.168.1: icmp_seq=666 ttl=64 time=8.329 ms
64 bytes from 192.168.168.1: icmp_seq=667 ttl=64 time=3.117 ms
Request timeout for icmp_seq 668
64 bytes from 192.168.168.1: icmp_seq=669 ttl=64 time=2.529 ms
64 bytes from 192.168.168.1: icmp_seq=670 ttl=64 time=1.881 ms
64 bytes from 192.168.168.1: icmp_seq=671 ttl=64 time=2.522 ms
64 bytes from 192.168.168.1: icmp_seq=672 ttl=64 time=1.523 ms
64 bytes from 192.168.168.1: icmp_seq=673 ttl=64 time=9.509 ms
64 bytes from 192.168.168.1: icmp_seq=674 ttl=64 time=9.758 ms
64 bytes from 192.168.168.1: icmp_seq=675 ttl=64 time=3.261 ms
64 bytes from 192.168.168.1: icmp_seq=676 ttl=64 time=9.246 ms
64 bytes from 192.168.168.1: icmp_seq=677 ttl=64 time=15.283 ms
64 bytes from 192.168.168.1: icmp_seq=678 ttl=64 time=71.470 ms
64 bytes from 192.168.168.1: icmp_seq=679 ttl=64 time=3.129 ms
64 bytes from 192.168.168.1: icmp_seq=680 ttl=64 time=6.796 ms

greearb, 2021-11-29:
that could be perfectly legit if station is in power save mode, and anyway, unless you are confident this is some bug in my ath10k, then not sure it is worth a bug?

gsustek, 2021-11-30:
i will close it and it should stay for documentation purposes.:-)
do not forget to switch to "performance" cpu governor.

`for i in {0..7}; do echo performance > /sys/devices/system/cpu/cpufreq/policy$i/scaling_governor; done
for i in {0..7}; do cat /sys/devices/system/cpu/cpu$i/cpufreq/scaling_governor; done`

closed 2021-11-30 by gsustek