greearb/ath10k-ct #198: Ath10K-CT With WLE1216V5-20 5Ghz with QCA9984

nashford77, 2022-02-13:
Ubuntu 20.04.03LTS 
 5.4.0-99-generic #112 SMP

ath10k_pci 0000:07:00.0: kconfig debug 1 debugfs 1 tracing 1 dfs 1 testmode 0
driver=ath10k_pci driverversion=5.4.0-99-generic firmware=10.4-3.9.0.2-00131

All - I am trying (without success) to compile this for Ubuntu X64 - First Question is - How do i do this as the instructions are not clear and LEDE base only for OpenWRT - I am making a beast router and that won't cut it on a few fronts - I need X64

Once I moved to CT FW, it disabled all the useful frequencies :( I can't even get them back now trying to restore the Stock FW - what happened ? This is NOT how it was pre - upgrade, it had all bands pretty much with IR & DFS which is ok - I upgraded to make the card understand the DFS results etc & poof - all gone now.... 

                Frequencies:
                        * 5180 MHz [36] (23.0 dBm) (no IR)
                        * 5200 MHz [40] (23.0 dBm) (no IR)
                        * 5220 MHz [44] (23.0 dBm) (no IR)
                        * 5240 MHz [48] (23.0 dBm) (no IR)
                        * 5260 MHz [52] (23.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (23.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (23.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (23.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (disabled)
                        * 5520 MHz [104] (disabled)
                        * 5540 MHz [108] (disabled)
                        * 5560 MHz [112] (disabled)
                        * 5580 MHz [116] (disabled)
                        * 5600 MHz [120] (disabled)
                        * 5620 MHz [124] (disabled)
                        * 5640 MHz [128] (disabled)
                        * 5660 MHz [132] (disabled)
                        * 5680 MHz [136] (disabled)
                        * 5700 MHz [140] (disabled)
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (30.0 dBm) (no IR)
                        * 5765 MHz [153] (30.0 dBm) (no IR)
                        * 5785 MHz [157] (30.0 dBm) (no IR)
                        * 5805 MHz [161] (30.0 dBm) (no IR)
                        * 5825 MHz [165] (30.0 dBm) (no IR)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)

My Q is : How do i correct the bands, whats a working config for this card and X86 ref FW-5. board2 files to fix plus how do i use the CT driver with Ubuntu ? 

Many Thanks in advance.

greearb, 2022-02-14:
You can just compile and use our pre-patched kernel:
https://github.com/greearb/linux-ct-5.15

nashford77, 2022-02-14:
Will this correct the fact that the CT FW disabled my bands that were on before I upgraded the cards FW ?

greearb, 2022-02-14:
I don't really know what you did, but the ath10k-ct driver with ath10k-ct firmware works fine with those channels if your regulatory domain allows it.

nashford77, 2022-02-14:
It does - it's the US. I upgraded the FW to the most current CT version,
used the stock board file, it's shows fine but before I could see all the
bands as NO IR & Radar which was fine then on reboot with ath10k driver
still, all the same channels went to disabled ?

I am trying to use channel 100 VHT160 to be clear which this card can do,
something is setting them to "disabled" now . Not sure what . I tried the
listed board file that CT team said you may need, no go - shows no FW
version then when I try that board file.

That's all I changed to date , FW, stock board file I reverted to, shows
channels disabled. What could cause this when it says US Reg ?

greearb, 2022-02-14:
Please show output of 'iw reg get'

And, try setting regdom to US using fwcfg API:

http://www.candelatech.com/ath10k-10.4.php#config

nashford77, 2022-02-14:
iw reg set US

country US: DFS-FCC
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
        
        working on the other item now ;0

nashford77, 2022-02-14:
I assume fwcfg is only available in your kernel  ? 

cat: /debug/ieee80211/wiphy3/ath10k/firmware_info: No such file or directory

Does this work on stock Ubuntu 20.04.03 LTS by chance ? If not I can install the Kernel you send me and try that

greearb, 2022-02-14:
fwcfg is only in our kernel, and you can configure the kernel options as you deem prudent.  Our kernel should work fine on Ubuntu, though we do not actually test much on that OS.

nashford77, 2022-02-14:
ah ok got it - so the Driver not the FW is looking for this file on boot to correct my messed up REG? Will try to compile your kernel on Ubuntu and see what it nets me to correct this issue.

nashford77, 2022-02-15:
Question: As I am on Kernel 5.4.x - could I not just do the following:

1) Copy the contents of this into my current ath10K driver for a custom kernel I already modified?
https://github.com/greearb/ath10k-ct/tree/master/ath10k-5.4 -->~/custom-kernel/linux-5.4.0/drivers/net/wireless/ath/ath10k/

2) Will This add support for 'fwcfg' to the kernel or is there another step to add this in for a compile ? 

3) Will This version have VHT160 support  - not 80 + 80 ? 

4) Will custom files in the FW area allow bypass of messed up card Reg then ?

nashford77, 2022-02-20:
Am up and running - for others:

1) correct
2) incorrect - fwcfg is not there but it didnt matter as i did another work around similar to what OpenWRT does
3) it supports 160 & DFS
4) N/A now due to patching with other patches not considered main stream ;0

nashford77, 2022-02-20:
closing this - i moved past this issue but have others which are open in other tickets

closed 2022-02-20 by nashford77