greearb/ath10k-ct #200: VHT160 Connects but not usable Kernel 5.4.0.99 Ubuntu 20.04

nashford77, 2022-02-20:
[    7.654168] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[    7.654188] ath10k_mac_create, priv_size: 1624  hw: 0000000005a90f46  hw->priv: 00000000648ef469
[    7.658344] ath10k_pci 0000:07:00.0: enabling device (0000 -> 0002)
[    7.658850] ath10k_pci 0000:07:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[    7.771896] ath10k_pci 0000:07:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[    7.771898] ath10k_pci 0000:07:00.0: kconfig debug 1 debugfs 1 tracing 1 dfs 1 testmode 1
[    7.772686] ath10k_pci 0000:07:00.0: firmware ver 10.4b-ct-9984-fW-13-774502ee5 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 5ffda406
[    9.033972] ath10k_pci 0000:07:00.0: board_file api 2 bmi_id 0:31 crc32 85498734
[   11.494529] ath10k_pci 0000:07:00.0: unsupported HTC service id: 1536
[   11.494772] ath10k_pci 0000:07:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   11.494773] ath10k_pci 0000:07:00.0: msdu-desc: 2500  skid: 32
[   11.571955] ath10k_pci 0000:07:00.0: wmi print 'ChipType 1100 ChipTypeID 0 BoardID 2 P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'

[  137.165392] ath10k_pci 0000:07:00.0: wmi print 'free: 85084 iram: 18852 sram: 12824'
[  137.535083] ath10k_pci 0000:07:00.0: rts threshold -1
[  137.538234] ath10k_pci 0000:07:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  137.550024] ath10k_pci 0000:07:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[  199.309101] ath10k_pci 0000:07:00.0: mac flush null vif, drop 0 queues 0xffff
[  201.840094] ath10k_pci 0000:07:00.0: wmi: fixing invalid VHT TX rate code 0xff
[  369.308840] ath10k_pci 0000:07:00.0: Invalid VHT mcs 15 peer stats
[ 2387.546870] ath10k_pci 0000:07:00.0: htt tx: fixing invalid VHT TX rate code 0xff
[21183.724108] ath10k_pci 0000:07:00.0: NIC rx-max-rate: 0 calculated-max: 0 rxnss_override: 0x80000000  nss160: 1  spatial-streams: 2
[21290.372624] ath10k_pci 0000:07:00.0: NIC rx-max-rate: 0 calculated-max: 0 rxnss_override: 0x80000000  nss160: 1  spatial-streams: 2
[22138.840208] ath10k_pci 0000:07:00.0: NIC rx-max-rate: 0 calculated-max: 0 rxnss_override: 0x80000000  nss160: 1  spatial-streams: 2
[22155.069078] ath10k_pci 0000:07:00.0: NIC rx-max-rate: 0 calculated-max: 0 rxnss_override: 0x80000000  nss160: 1  spatial-streams: 2
[22222.461250] ath10k_pci 0000:07:00.0: NIC rx-max-rate: 0 calculated-max: 0 rxnss_override: 0x80000000  nss160: 1  spatial-streams: 2

nashford77, 2022-02-20:
Also, why do i get :: unsupported HTC service id: 1536

nashford77, 2022-02-20:
Also - the few times it does connect, the ping is horrid .... 

PING 192.168.5.135 (192.168.5.135) 56(84) bytes of data.
64 bytes from 192.168.5.135: icmp_seq=1 ttl=64 time=45.8 ms
64 bytes from 192.168.5.135: icmp_seq=2 ttl=64 time=68.3 ms
64 bytes from 192.168.5.135: icmp_seq=3 ttl=64 time=91.0 ms
64 bytes from 192.168.5.135: icmp_seq=4 ttl=64 time=113 ms
64 bytes from 192.168.5.135: icmp_seq=5 ttl=64 time=35.1 ms
64 bytes from 192.168.5.135: icmp_seq=6 ttl=64 time=2.63 ms
64 bytes from 192.168.5.135: icmp_seq=7 ttl=64 time=79.2 ms
64 bytes from 192.168.5.135: icmp_seq=8 ttl=64 time=102 ms
64 bytes from 192.168.5.135: icmp_seq=9 ttl=64 time=22.3 ms
 
 Please advise @greearb when you have time - many thanks.

nashford77, 2022-02-20:
To add more info, anything client wise with a 160MHZ radio wont work - also, 80+80 says not supported.... ? ;0 #help ;0

nashford77, 2022-02-20:
I have perf governor set - lots of issues.... latency and some clients just don't work ;0 #Help 

for i in {0..15}; do cat /sys/devices/system/cpu/cpu$i/cpufreq/scaling_governor; done
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance
performance

ping 192.168.5.135
PING 192.168.5.135 (192.168.5.135) 56(84) bytes of data.
64 bytes from 192.168.5.135: icmp_seq=1 ttl=64 time=32.8 ms
64 bytes from 192.168.5.135: icmp_seq=2 ttl=64 time=43.6 ms
64 bytes from 192.168.5.135: icmp_seq=3 ttl=64 time=66.8 ms
64 bytes from 192.168.5.135: icmp_seq=4 ttl=64 time=88.7 ms
64 bytes from 192.168.5.135: icmp_seq=5 ttl=64 time=213 ms
64 bytes from 192.168.5.135: icmp_seq=6 ttl=64 time=134 ms
64 bytes from 192.168.5.135: icmp_seq=7 ttl=64 time=63.7 ms
64 bytes from 192.168.5.135: icmp_seq=8 ttl=64 time=85.2 ms
64 bytes from 192.168.5.135: icmp_seq=9 ttl=64 time=107 ms
64 bytes from 192.168.5.135: icmp_seq=10 ttl=64 time=29.4 ms
64 bytes from 192.168.5.135: icmp_seq=11 ttl=64 time=50.9 ms
64 bytes from 192.168.5.135: icmp_seq=12 ttl=64 time=64.5 ms
^C
--- 192.168.5.135 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 11017ms
rtt min/avg/max/mdev = 29.351/81.615/212.920/49.176 ms

nashford77, 2022-02-20:
That is from an android cell connected 5 feet away from the AP - line of sight, 160 MHZ channel 100. Something is off

nashford77, 2022-02-20:
Station 64:bc:58:b9:1e:eb (on wlp7s0)
        inactive time:  1240 ms
        rx bytes:       20453364
        rx packets:     110408
        tx bytes:       3454212
        tx packets:     20291
        tx retries:     26
        tx failed:      41
        rx drop misc:   2523
        signal:         -76 [-78, -79] dBm
        signal avg:     -75 [-77, -78] dBm
        tx bitrate:     162.0 MBit/s VHT-MCS 4 40MHz VHT-NSS 2
        tx duration:    377394 us
        rx bitrate:     526.5 MBit/s VHT-MCS 6 160MHz VHT-NSS 1
        rx duration:    10376772 us
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 35759 seconds
        associated at [boottime]:       33510.624s
        associated at:  1645343523357 ms
        current time:   1645379281905 ms
Station 42:30:92:08:5d:16 (on wlp7s0)
        inactive time:  200 ms
        rx bytes:       586522
        rx packets:     2769
        tx bytes:       452353
        tx packets:     1492
        tx retries:     1
        tx failed:      10
        rx drop misc:   0
        signal:         -66 [-96, -96, -66] dBm
        signal avg:     -66 [-86, -87, -67] dBm
        tx bitrate:     433.3 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 1
        tx duration:    36555 us
        rx bitrate:     6.0 MBit/s
        rx duration:    282916 us
        airtime weight: 256
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 931 seconds
        associated at [boottime]:       68338.803s
        associated at:  1645378351535 ms
        current time:   1645379281905 ms

Also why am i receiving at VHT160 and sending on another speed ? ;0

gsustek, 2022-02-21:
you should update your kernel/other packages with some openwrt patches:
https://forum.openwrt.org/t/migrate-x86-64-openwrt-patches-to-linux-distro/114853

greearb, 2022-02-21:
the ping latency is caused by power-save, it is a feature.  The station dump shows weak signal (-76), that would explain relativley low encoding rates.  If that station dump with weak RSSI is from a close distance, then you are missing antenna connectors or have some other bad flaw with your setup.  Possibly a bad board file for whatever NIC you are using.

nashford77, 2022-02-21:
What causes the VHT160 radio to connect @160, get an IP then not work at
all nor be pingable ?

Ref to physical setup, I will recheck but all leads were connected etc - I
would be surprised if there's an issue but it's possible. Will revert on
that.

nashford77, 2022-02-21:
@greearb No matter what i try in Ubuntu it tells me the driver doesn't support to disable WiFi power ? 

iw wlp7s0 set power_save off
command failed: Operation not supported (-95)

i dont see a PS option in the kernel module.

filename:       /lib/modules/5.4.0-99-generic/kernel/drivers/net/wireless/ath/ath10k/ath10k_core.ko
license:        Dual BSD/GPL
description:    Core module for Qualcomm Atheros 802.11ac wireless LAN cards.
author:         Qualcomm Atheros
srcversion:     79A66213A89D6DB4B6E2299
depends:        mac80211,cfg80211,ath
retpoline:      Y
intree:         Y
name:           ath10k_core
vermagic:       5.4.0-99-generic SMP mod_unload modversions
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        24:27:4D:D0:A6:D1:DB:E3:40:03:DE:11:E7:03:3C:6F:28:8D:33:95
sig_hashalgo:   sha512
signature:      31:25:B6:51:8C:33:2E:49:9B:D1:6A:D3:7B:18:36:D2:7B:5E:70:3A:
                73:2E:AE:6E:2C:BF:B2:DA:94:1E:A6:C5:F8:B0:68:88:C3:EF:1B:4A:
                FD:5C:49:BD:F9:F7:37:64:56:AB:CE:95:8D:1A:D6:FD:3F:21:A9:FD:
                F2:DA:C0:3A:8E:1D:E3:27:75:5A:76:60:91:03:68:CD:F5:E4:7C:F5:
                2D:73:4F:52:59:75:16:FF:68:A6:BF:74:4B:29:A6:C6:04:25:7B:25:
                71:56:9A:90:8B:95:0D:60:58:E7:8D:1D:EB:3F:5B:39:34:A6:0E:DE:
                44:43:AF:78:B1:66:59:1E:1C:6D:A8:BE:27:E8:41:98:0A:F0:62:C6:
                26:3D:CC:7C:ED:5E:99:DD:83:A7:67:F0:40:82:C8:A2:2C:72:AC:FA:
                58:CC:A2:D9:6C:59:DD:4E:8C:75:7C:0D:0D:A7:4A:DD:08:AD:C2:CA:
                D4:09:8C:FE:B4:90:71:81:43:47:CB:00:96:4E:08:35:14:B8:DB:9A:
                01:7B:38:0F:9D:22:50:E1:48:91:B6:DA:6A:74:5F:64:FB:96:AB:5A:
                DC:56:37:F3:C1:23:BF:F5:56:65:6B:4A:69:96:4B:84:40:4F:2C:7F:
                A8:58:61:A1:BD:DC:D6:E7:9C:64:36:DB:4F:9D:B5:90:C3:AF:A6:C4:
                98:77:AD:41:32:33:41:14:47:BE:23:C1:1C:43:4B:7A:95:C7:69:EA:
                88:2A:DC:DE:15:81:34:FF:85:C8:CF:EA:9A:6C:6B:82:10:89:85:23:
                B9:E6:1B:EA:98:71:4D:94:68:F0:F8:E0:8A:A2:E4:98:0F:43:20:00:
                05:32:7B:23:CB:89:E3:4A:C9:ED:E3:8D:A4:87:62:C1:94:18:A2:E2:
                08:47:05:E8:7E:CF:06:83:1A:EA:8C:C7:D2:AB:CE:B1:E1:9A:E6:A7:
                A8:C4:64:23:09:B9:18:01:9C:25:31:EA:BD:54:5B:EC:5E:4B:80:54:
                C8:B3:DC:40:C1:69:2F:93:53:15:05:F4:F8:E9:AE:86:C5:D6:27:71:
                91:2D:91:70:B3:20:6C:21:A4:9F:20:1E:30:70:59:22:E7:01:D4:06:
                9E:53:EF:B1:A0:76:E7:0E:BE:BA:DA:DA:E1:8A:86:86:46:98:75:04:
                95:EA:0B:08:93:2F:82:DD:72:2E:5E:4F:9D:74:B1:22:18:11:3D:56:
                85:B9:17:DB:F6:7F:4D:EC:36:A8:B6:A1:CF:75:3B:87:21:7C:F1:08:
                91:5C:22:B4:D2:A7:BC:55:B7:9A:C0:E2:71:54:1A:01:A5:E1:FD:F8:
                E0:E3:9D:4F:51:6D:B0:E0:A6:3A:F1:87
parm:           override_eeprom_regdomain:Override regdomain hardcoded in EEPROM with this value (DANGEROUS). (int)
parm:           debug_mask:Debugging mask (uint)
parm:           uart_print:Uart target debugging (bool)
parm:           skip_otp:Skip otp failure for calibration in testmode (bool)
parm:           cryptmode:Crypto mode: 0-hardware, 1-software (uint)
parm:           rawmode:Use raw 802.11 frame datapath (bool)
parm:           coredump_mask:Bitfield of what to include in firmware crash file (ulong)
parm:           fw_diag_log:Diag based fw log debugging (bool)
parm:           nohwcrypt:Disable hardware rx decrypt feature (int)
parm:           ct_sta:Use CT-STA mode, a bit like proxy-STA (int)
parm:           nobeamform_mu:Disable TX/RX MU Beamforming capabilities (int)
parm:           nobeamform_su:Disable TX/RX SU Beamforming capabilities (int)
parm:           num_vdevs_ct:Maximum vdevs to request from firmware (int)
parm:           num_peers_ct:Maximum peers to request from firmware (int)
parm:           num_msdu_desc_ct:Maximum MSDU Descriptors in firmware (must be multiple of 8) (int)
parm:           num_rate_ctrl_objs_ct:Number of rate-ctrl objects to cache in firmware RAM (int)

gsustek, 2022-02-24:
gs@u1804:~$ sudo iw wlp7s0 set power_save off
gs@u1804:~$ sudo iw wlp7s0 get power_save
Power save: off

nashford77, 2022-02-24:
I am on 20.04 - no mods except the ones from OpenWRT to the kernel, was
stock prior. Can't turn it off for the life of me.