greearb/ath10k-ct #201: Ubuntu 20.04 Compile 5.15.24 Kernel - Compile Error

nashford77, 2022-02-21:
You do not need the "sizeof(ar->mac_addr)" third function here - i removed it and it compiled fine:

drivers/net/wireless/ath/ath10k/core.c: In function ‘ath10k_core_probe_fw’:
drivers/net/wireless/ath/ath10k/core.c:4049:2: error: too many arguments to function ‘device_get_mac_address’
 4049 |  device_get_mac_address(ar->dev, ar->mac_addr, sizeof(ar->mac_addr));
      |  ^~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/ieee80211.h:20,
                 from ./include/net/mac80211.h:20,
                 from drivers/net/wireless/ath/ath10k/htt.h:16,
                 from drivers/net/wireless/ath/ath10k/core.h:18,
                 from drivers/net/wireless/ath/ath10k/core.c:19:
./include/linux/etherdevice.h:35:5: note: declared here
   35 | int device_get_mac_address(struct device *dev, char *addr);
      |     ^~~~~~~~~~~~~~~~~~~~~~

greearb, 2022-02-21:
It must be conflicting with some patch that ubuntu pulls into their kernel that is not in mainline 5.15 kernel,
as my 5.15 kernel compiles for me...

nashford77, 2022-02-21:
Also ref to power save - I set performance governor on all cores - didn't
make a diff. Still the same.

greearb, 2022-02-21:
You need to google 'wifi power save' and read up a bit, it has nothing
to do with CPU power save/performance.

I don't know why you have issues with 160Mhz.  Does 80Mhz work better?