greearb/ath10k-ct #207: QCA9984 2.4G AP mode no beacon

NathanKuo0821, 2022-09-06:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
We use QCA9984 2.4G as AP mode on PowerPC with CT driver and CT firmware. When the AP starts up, we will find that the AP does not send any beacon, and we will see from the firmware status that the PDEV reset counter keeps increasing. Any idea why this problem occurs?

Software (OS, Firmware version, kernel, driver, etc)
Firmware version:10.4b-ct-9984-fH-13-774502ee5
Driver :  backports.git v5.7.5-1-0-g34b56f34
Kerenl : 4.1.35-rt41-

Hardware (NIC chipset, platform, etc)
PowerPC

Logs (dmesg, maybe supplicant and/or hostap)
[hostapd_log.txt](https://github.com/greearb/ath10k-ct/files/9493515/hostapd_log.txt)
[firmware_status_log.txt](https://github.com/greearb/ath10k-ct/files/9493516/firmware_status_log.txt)
[hostapd_conf.txt](https://github.com/greearb/ath10k-ct/files/9493517/hostapd_conf.txt)
[ath10k_init_log.txt](https://github.com/greearb/ath10k-ct/files/9493518/ath10k_init_log.txt)
[nework_config.txt](https://github.com/greearb/ath10k-ct/files/9493519/nework_config.txt)

greearb, 2022-09-06:
Does your system work if you use upstream driver and firmware?

NathanKuo0821, 2022-09-13:
Hi Greearb,

We used the upstream driver and ath10k firmware 'https://github.com/kvalo/ath10k-firmware/tree/master/QCA9984/hw1.0' and then the issue still exists , Do you think this is hardware RF or software issue? Can you give me some advice? 
Very thanks.

Nathan Kuo

greearb, 2022-10-11:
Try using known-working x86-64 hardware so that you can at least debug your basic
configuration, then once that is working, you can try your specific board.

Thanks,
Ben