greearb/ath10k-ct #208: bug Since release

Oozil, 2023-01-07:
I was able to open 4.9GHz channels and 5.9GHz channels on ATH10K-4.19 but the latest versions did not work and do not appear in the openWRT system interface such as ATH10K-4.19 only appear through this command putty ssh ( iw phy
How can I fix that on the latest releases

lemoer, 2023-01-13:
You can change the following lines:

greearb, 2023-01-13:
The ath10k-ct driver versions need to match the backports, unless you have skill and willingness to effectively back/forward port the driver to work with whatever backports you are using.  Also, the change in channel support might be from elsewhere in mac80211, not just ath10k.  I did remove some out-of-tree hacks related to 4.9 from my kernel, not sure if that affected ath10k driver much or not.

Oozil, 2023-01-13:
> تحتاج إصدارات برنامج التشغيل ath10k-ct إلى مطابقة المنافذ الخلفية ، ما لم تكن لديك المهارة والرغبة في إعادة / توجيه السائق بشكل فعال للعمل مع أي منفذ خلفي تستخدمه. أيضًا ، قد يكون التغيير في دعم القناة من مكان آخر في mac80211 ، وليس فقط ath10k. لقد قمت بإزالة بعض الاختراقات خارج الشجرة المتعلقة بـ 4.9 من نواة بلدي ، ولست متأكدًا مما إذا كان ذلك قد أثر على برنامج تشغيل ath10k كثيرًا أم لا.

Indeed, this modification affected ath10k, and I want to compile ath10k-4.19 with the latest commit from openwrt, but it gives me an error in the compilation only. It works with ath10k-5.15. How can I fix that? 
https://github.com/openwrt/openwrt/commit/9c0c1c44013003616e2f5c55c062430dbda3cce5

Oozil, 2023-01-13:
> The ath10k-ct driver versions need to match the backports, unless you have skill and willingness to effectively back/forward port the driver to work with whatever backports you are using. Also, the [...]

In file included from <command-line>:
drivers/usb/core/hub.c: In function 'hub_configure':
/home/Oozel/Pictures/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/backport/backport.h:10:31: error: too many arguments to function 'backport_usb_maxpacket'
   10 | #define LINUX_BACKPORT(__sym) backport_ ##__sym
      |                               ^~~~~~~~~
/home/Oozel/Pictures/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/usb.h:11:23: note: in expansion of macro 'LINUX_BACKPORT'
   11 | #define usb_maxpacket LINUX_BACKPORT(usb_maxpacket)
      |                       ^~~~~~~~~~~~~~
drivers/usb/core/hub.c:1638:16: note: in expansion of macro 'usb_maxpacket'
 1638 |         maxp = usb_maxpacket(hdev, pipe, usb_pipeout(pipe));
      |                ^~~~~~~~~~~~~
In file included from drivers/usb/core/hub.c:23:
/home/Oozel/Pictures/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/linux/usb.h:7:19: note: declared here
    7 | static inline u16 backport_usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:289: drivers/usb/core/hub.o] Error 1
make[7]: *** Waiting for unfinished jobs....
  CC [M]  net/ipv6/netfilter/nf_flow_table_ipv6.o
  CC [M]  net/netfilter/nf_nat_core.o
  CC [M]  net/netfilter/nf_nat_proto.o
  CC [M]  net/netfilter/nf_nat_helper.o
  CC [M]  net/netfilter/nf_nat_redirect.o
drivers/usb/core/message.c:2284:33: warning: 'struct usb_cdc_parsed_header' declared inside parameter list will not be visible outside of this definition or declaration
 2284 | int cdc_parse_cdc_header(struct usb_cdc_parsed_header *hdr,
      |                                 ^~~~~~~~~~~~~~~~~~~~~
drivers/usb/core/message.c: In function 'cdc_parse_cdc_header':
drivers/usb/core/message.c:2301:34: error: invalid application of 'sizeof' to incomplete type 'struct usb_cdc_parsed_header'
 2301 |         memset(hdr, 0x00, sizeof(struct usb_cdc_parsed_header));
      |                                  ^~~~~~
drivers/usb/core/message.c:2302:12: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2302 |         hdr->phonet_magic_present = false;
      |            ^~
drivers/usb/core/message.c:2332:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2332 |                         hdr->usb_cdc_country_functional_desc =
      |                            ^~
drivers/usb/core/message.c:2345:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2345 |                         hdr->usb_cdc_acm_descriptor =
      |                            ^~
drivers/usb/core/message.c:2358:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2358 |                         hdr->usb_cdc_call_mgmt_descriptor =
      |                            ^~
drivers/usb/core/message.c:2364:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2364 |                         hdr->usb_cdc_dmm_desc =
      |                            ^~
drivers/usb/core/message.c:2384:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2384 |                         hdr->usb_cdc_ncm_desc = (struct usb_cdc_ncm_desc *)buffer;
      |                            ^~
drivers/usb/core/message.c:2390:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2390 |                         hdr->usb_cdc_mbim_desc = (struct usb_cdc_mbim_desc *)buffer;
      |                            ^~
drivers/usb/core/message.c:2395:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2395 |                         hdr->usb_cdc_mbim_extended_desc =
      |                            ^~
drivers/usb/core/message.c:2398:22: error: 'CDC_PHONET_MAGIC_NUMBER' undeclared (first use in this function)
 2398 |                 case CDC_PHONET_MAGIC_NUMBER:
      |                      ^~~~~~~~~~~~~~~~~~~~~~~
drivers/usb/core/message.c:2398:22: note: each undeclared identifier is reported only once for each function it appears in
drivers/usb/core/message.c:2399:28: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2399 |                         hdr->phonet_magic_present = true;
      |                            ^~
drivers/usb/core/message.c:2415:12: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2415 |         hdr->usb_cdc_union_desc = union_header;
      |            ^~
drivers/usb/core/message.c:2416:12: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2416 |         hdr->usb_cdc_header_desc = header;
      |            ^~
drivers/usb/core/message.c:2417:12: error: invalid use of undefined type 'struct usb_cdc_parsed_header'
 2417 |         hdr->usb_cdc_mdlm_detail_desc = detail;
[... 46 lines cut ...]
  CC [M]  net/netfilter/nft_log.o
  CC [M]  net/netfilter/nft_masq.o
  CC [M]  net/netfilter/nft_redir.o
  CC [M]  net/netfilter/nft_hash.o
  CC [M]  net/netfilter/nft_fib.o
  CC [M]  net/netfilter/nft_fib_inet.o
  CC [M]  net/netfilter/nft_chain_nat.o
  CC [M]  net/netfilter/nf_flow_table_core.o
  CC [M]  net/netfilter/nf_flow_table_ip.o
  CC [M]  net/netfilter/nf_flow_table_offload.o
  CC [M]  net/netfilter/nf_flow_table_inet.o
[... cut ...]
time: package/kernel/ath10k-ct/regular/compile#169.53#25.84#29.77
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/home/Oozel/Pictures/openwrt'
make[1]: *** [package/Makefile:110: /home/Oozel/Pictures/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/Oozel/Pictures/openwrt'
make: *** [/home/Oozel/Pictures/openwrt/include/toplevel.mk:231: world] Error 2``