greearb/ath10k-ct #212: ATH10K Archer C7v2 running CT-full-HTT

UltraHKR, 2023-04-02:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
I recently upgraded from old 21.02 Snapshot to lastest Snapshot, and the AP is FW crashing very fast every 120-140 secs or so...
I only have 2 or 3 devices connected to this AP

Disabled 802.11w (was setup in Optional mode) rebooted, and its stable....

Software (OS, Firmware version, kernel, driver, etc)
Openwrt "openwrt-57f37afa866d-ath79-generic-tplink_archer-c7-v2-squashfs-factory.bin"
DAWN v2022-07-24-9e8060ea-3

Hardware (NIC chipset, platform, etc)
Tp-Link Archer C7 v2

Logs (dmesg, maybe supplicant and/or hostap)
[ath10k dmesg.txt](https://github.com/greearb/ath10k-ct/files/11131140/ath10k.dmesg.txt)

UltraHKR, 2023-04-02:
Added a long dmesg, without 802.11w is stable, enable it and FW goes bonkers
[ath10k dmesg not ok.txt](https://github.com/greearb/ath10k-ct/files/11131165/ath10k.dmesg.not.ok.txt)

timocapa, 2023-04-19:
Seems the same as https://github.com/greearb/ath10k-ct/issues/206

Don't think the developer has ideas

mixman68, 2023-12-20:
Hello

i just see the topic after i reported something on OpenWRT forum

so you have this option ?  
option ieee80211k '1'

I don't know why, this option was the resolution of the crash with DAWN or usteer installed